probonopd/RF100-Firmware #15: Allow manual fan speed control

krisrok, 2018-04-01:
Maybe there is a way? I'm not familiar around the marlin codebase.

probonopd, 2018-04-01:
Have you tried M106 in your GCODE?
http://marlinfw.org/docs/gcode/M106.html

probonopd, 2018-04-01:
Possibly can change something around

https://github.com/probonopd/RF100-Firmware/blob/d7f264d6506210bc4afd0592230394e4700a00bc/RF-100/Configuration_adv.h#L203-L221

What exactly do you want to achieve?

krisrok, 2018-04-01:
Ah, should have clarified: I want to be able to tune the cooling fan speed via LCD/knob during a print. Like it was in the original firmware.

From my quick glance over the sources it looks like it _may_ conflict with the auto fan behaviour you mentioned.

probonopd, 2018-04-01:
Good question; I don't know. Maybe ask in https://github.com/MarlinFirmware/Marlin

krisrok, 2018-04-02:
Referencing https://github.com/MarlinFirmware/Marlin/issues/10280

I guess that's why you manually set 
`  #define FAN_PIN        -1 //RAMPS_D9_PIN`
in pins_RAMPS.h

probonopd, 2018-04-03:
Without any guarantees, what happens if you set 

```
#undef FAN_PIN
#define FAN_PIN           9    // FAN
```
?

It is mentioned in https://github.com/MarlinFirmware/Marlin/issues/2906#issue-127145428 about the mpx.3 which seems to be a similar board from what is inside the RF100. But I assume no responsibility if it blows up your printer.

probonopd, 2018-04-06:
I can reproduce that with this firmware we currently cannot control the speed of the fan. Any help appreciated on how to get that feature back,

krisrok, 2018-04-08:
I'll work on it starting next week i guess. Really missing the feature.

probonopd, 2018-10-13:
@krisrok did you find something out?

probonopd, 2018-10-13:
It is indeed very annoying that the fan runs at full speed all the time. I think it is far too much, especially since the fan blows inward against a wall, and the air then bounces back outward... not ideal

krisrok, 2018-10-13:
hey, indeed. i had introduced a parameter to act as a multiplier for the fan speed. sadly it used to stop printing because of temperature deviations and could not figure out the source of it.

(i went back to your version and also had the stops because of temps. then i went back to the renkforce firmware and had no more temperature related halts. some time later i noticed my temperatuer sensor went south. currently i'm waiting for a replacement from conrad since about 2 month.)

so i could commit my changes (beware: incomplete) and you can review it / try it on your printer / finish the feature?

krisrok, 2018-10-13:
nah sorry, i must have discarded my local repository during a hdd change.

krisrok, 2018-10-13:
anyway. found it: https://github.com/krisrok/RF100-Firmware/commit/872db998af2bb0b2c5a0d67fbd5847584063e881

i just don't think it's of much use :( ;)

probonopd, 2018-10-13:
Thank you very much. Did you invent `fanUserAutoMultiplier`?

krisrok, 2018-10-13:
yes, the rationale was to use it as a factor to before the actual fan speed is set. so you could adjust the speed while printing.

probonopd, 2018-10-13:
I wonder what would happen if we just globally reduced the speed to, say, half.

probonopd, 2018-10-13:
Much better I'd say.

krisrok, 2018-10-13:
for a quick noise cancelling fix it sure is an easy option :)

there are times where you need good steady airflow though. especially in combination with a part cooling like this: https://www.thingiverse.com/thing:2057291

really helps with printing details and bridging.